KunaiGC/KunaiGC #5: debugging install

Colrut, 2023-05-18:
good morning

I am trying to debug an install and had two questions

1 - please check this continuity map.  the colors are a bit hard to see so I numbered them.  point 2 does not seem to connect anyplace in the source picture.

2 - is there a guide for the flashing lights?  how do we use these do check an install.

thank you

[image]

Colrut, 2023-05-18:
[image]

better picture

ManCloud, 2023-05-18:
The pin 2 is on the picture below (in the wiki... Other side of the mainboard).

If everything is okay, both LEDs do something. If only one led is doing something most probably the exit.out line is defective.

If no led does anything, anything might be defective.

ManCloud, 2023-05-18:
EXI.CLK = your Nr. 7 ^^ 
It's pin #1 on the IPL chip.

Colrut, 2023-05-19:
Good morning

 

Thank you for your response and assistance.  Here are the steps that I have taken

 

1.	Checked/reflow all solder joints
2.	Checked continuity from chip to side of IPL, to IPL, and to the points on the KunaiGC itself
3.	Booted with and without the SD card with the special folder
4.	Held down d-pad down or start during boot

 

Each time light 1 will flash red and light 2 will then stay red.

It will always load regular gamecube menu – does not show the diagnostics or swiss

 

Any other suggestions?  Any chance of a bad or not programmed KunaiGC?

 

Thank you

 

Russell

Colrut, 2023-05-26:
[KunaiContinuity.pdf](https://github.com/KunaiGC/KunaiGC/files/11579085/KunaiContinuity.pdf)

I numbered the continuity to make it easier (at least for myself).  Feel free to share.
I likely had a bad board.  Tried direct wiring to the IPL with no benefit.  
I installed a different QSB and Kunai in the same cube and it works...

However none of the expected buttons on controller (z, start, d-down) have any function - it always boots to regular gamecube menu.  I checked the controller port and controller.  But if I put a SD card in with the KUNAI folder it autoboots to Swiss from here.  This does not match with the expected behavior.

Any suggestions?

Colrut, 2023-05-28:
Is there a .DOL version of recovery that can be run to help troubleshoot?

indrora, 2023-06-06:
I'm having the same issue. 

I got my KunaiGC from Consoles4you, verified all the connections, and the LEDs light as they should (momentary D1 -> solid D2).

No button presses activate anything: Holding start or Z simply boots straight to the normal Gamecube IPL.

ManCloud, 2023-06-06:
I was about to write that 3rd party controller stuff ^^

The embedded recovery unfortunately should not be touched... But can...
This would take the risk of bricking the KunaiGC.

So for recovery stuff it seems there always needs to be an OEM controller connected.

But for the normal operation it should definitely be possible to do so.

I have no 3rd party controllers here but could test with blueretro 👌

indrora, 2023-06-06:
I would absolutely suggest testing against these specific third party controllers:

* the ones from [CirKa](https://www.amazon.com/CirKa-Wired-Controller-GameCube-Wii-Purple/dp/B000UQCER6), which are very popular in shops here and which are likely based on the most common of clone designs
* BlueRetro (either external or the Laser Bear Industries internal mod)
* the Retro Fighters [wireless gamecube controller](https://retrofighters.com/our-collection/battlergc-next-gen-wireless-gamecube-controller/)
* a PhobGCC 

BlueRetro and PhobGCC being open source makes things easier -- the current PhobGCC is built around an RP2040 and the BlueRetro is around an esp32. If PhobGCC and BlueRetro can't activate the debug or built-in recovery, that's a bad problem.

ManCloud, 2023-06-15:
In my opinion there is no way to implement support for wireless controllers.
It would make the overall booting process very slow as we can never be sure a wireless controller is connected during boot especially when it comes to bluetooth controllers.

The internal Laser-Bear BlueRetro board on the other hand should already work as it usually is powered before the console itself is.

i'll see what i can do regarding the other controllers 👍

TharathielCB, 2023-11-23:
@Colrut: I added your suggestion to the wiki. https://github.com/KunaiGC/KunaiGC/wiki/Installation#testing

@indrora: We will look over the 3rd party Controller topic once we have some of them for testing here, but right now this is not the highest priority. Please open a separate issue if this topic is still present.

closed 2023-11-23 by TharathielCB