KunaiGC/KunaiGC #17: Clarity of wiki instructions for programming the CPLD and flash

Rhombic-Drive, 2024-01-06:
Hi,

I've currently built a few KunaiGCs and am ready to perform the flashing steps. The CPLD seems straghtforward, although I am not sure which jedec file I should be using to flash, as that is not mentioned in the wiki. The only IPL .jed file that I could find for the xc9536 was one bundled with the Shuriken Attack. Will this do the job for providing a basic IPL or should I be using something else?

I guess the wiki page on programming the flash memory is incomplete--I noticed the labels on the Raspberry Pi GPIO pins don't correspond to the SPI SOICbite pinouts exactly. Some clarification here would be helpful. I think the appropriate .bin file for the 
16Mx8 flash IC is KunaiGC_lfs_16M.bin.

Thank you

ManCloud, 2024-01-06:
First of all: the CPLD related stuff is in the KubaiGC-IPL Repository.
Second: what do you mean with "a few"...
Sounds like you're not building for personal use which would violate the licenses (non commercial use permitted)...
Hope that's not, what it means...

Rhombic-Drive, 2024-01-06:
I made 3. One is for myself, the other 2 are gifts. I'm not selling them.

I've seen the IPL repository; I didn't see a jed file but now that I look at the makefile I can see that it's generated there. Thanks. That would be good info to include in the wiki as well.

ManCloud, 2024-01-07:
Allright just wanted to make sure ^^
Unfortunately it happens way too often that people can't read license properly 🙈

We'll add that to the wiki, thanks for your input 👍

For the SPI part: i can see your confusion. This is because people tend to name the lines differently: some just call it DO (for data out) others call is MOSI (for Master Out Slave In). IIRC the picture of the RPI is referenced to the nomenclature used on "https://wiki.flashrom.org/RaspberryPi" as the schematics (and layout) uses the SPI nomenclature.

Usually this should not be a "big deal" for people dealing with this on a regular basis but can be quite confusing for people who don't.

we'll add a connection table in the near future 👍

Rhombic-Drive, 2024-01-07:
Thanks, that's helpful. I have programmed the CPLD successfully. I ended up using WSL to run the Linux version of ISE 14.7, since the Windows version was a headache. If you're using Ubuntu you need to add the universal repository to get a libncurses5 package. I tried openSUSE first, but the WSL X11 support wasn't working. Now I'm working on programming the flash but have a few questions:

1. Do the JTAG SOICbite connections still need to be maintained on the GPIO pins when connecting the SPI pins? I guess this is the case from what I've read on how SPI flash is programmed with these CPLDs.
2. For programming the SPI flash, the xc3sprog documentation mentions needing a bitfile (which isn't included by default in their `bscan_spi` folder. Do I need to use the Xilinx tools (Impact) approach here? In this case I'd have to figure that out first.

ManCloud, 2024-01-08:
The SPI-Flash needs to be programmed on beforehand (at least that is what I always did as in circuit programming did not work once the cpld was placed.

Therefore I used a tl866ii programmer or friends of mine used a raspberry pi for it.

If there is actually a way to upload it via jtag: I don't know.

And for cpld programming I used the Windows ISE (with virtual box first, then with fixes by gamebox systems) on win11.

Programmed cia impact.

Iirc it's the only way capable of erasing locked cpld therefore I myself never investigated in openocd stuff etc.

Rhombic-Drive, 2024-01-08:
From what I've seen the support isn't there in xc3sprog to make it easy when using the XC9536XL. I think going that route, or the ISE tools approach, is beyond my current skill level, so I will try using flashrom on my Pi first, since this W25 flash is supported. I will report back if this works, maybe that will be the easiest way for people who are already using their Pi for the CPLD.

Rhombic-Drive, 2024-01-16:
Unfortunately it's not, at least from what I can see. I desoldered the flash and will program it once my SOIC clip comes in. I'll report back after that. I can also write up my process for doing it if that would be helpful.

Rhombic-Drive, 2024-02-29:
Well, it took a while for me to get some decent QSB boards (needed to use PCBWay and re-order them--the JLCPCB ones were routed out way off target). I did end up programming the flash using flashrom on my Pi 3b. It appeared to have programmed successfully.

Unfortunately things aren't working from what I can tell. The solder joints are all good, and no neighboring pins on the QSB board are shorted. D1 lights up when the system is turned on. I am not currently getting any video signal out. I'm going to do some troubleshooting and make sure with a scope that I'm not getting the right signals out, in case there is just a problem with my OSSC getting sync or something.

I'll try to contribute to the wiki with my approach if I can get everything working.

olDirdey, 2024-05-24:
> Edit: After I removed the Kunai main board (QSB still there) it's working. So something may have gone wrong during programming, or it got ESD'd or something. I'll do some more troubleshooting.

which cpld did you install? if you take the source files and do not change anything, you will generate a firmware for the 9572. if you use the 9536 you have to change this before compiling.

Rhombic-Drive, 2024-05-24:
> which cpld did you install? if you take the source files and do not change anything, you will generate a firmware for the 9572. if you use the 9536 you have to change this before compiling.

I found this explanation in the Discord; I did change the firmware source file to 9536 and it worked after that. All's well.