KunaiGC/KunaiGC #18: "Update the KunaiGC" wiki page clarification

Paradox355, 2024-03-29:
Hello. I've been trying to update my KunaiGC to the latest KunaiLoader/KunaiMenu version but the wiki page is unclear on a few things.

Firstly, it said to add files to the KUNAIGC folder but there's no indication on _where_ that folder should reside (I'm assuming on the root of SD card).

Secondly, there's no information on comparing versions of the apps. My menu looks different than the screenshot, so I'm guessing I have the old version. But even the screenshot it unclear, the wiki page mentions KunaiLoader v1.1 but the screenshot shows KunaiUpdate v1.0. It's all very confusing.

Anyway here's what I attempted: Following the "Updating the Loader to Version 1.1 or higher" section, I placed two copies of KunaiLoader.dol from the latest release into KUNAIGC folder on SD card (it's formatted using FAT32 and I'm using SD2SP2) and I renamed one to recovery.dol. I put the SD2SP2 with SC card back into GameCube and powered on while holding Z. I just get the same old menu screen with no update process being kicked off. What am I doing wrong?

justinstrack, 2024-06-23:
I'm having the same issue following the instructions to the "letter" best I can but nothing seems to be working. 
[image]

CplMadMurdock, 2026-10-18:
I'm able to run the updater, but after a reboot it's still on 1.0. It doesn't seem to want to update to 1.1.